OpenRoberta/openroberta-lab #1464: Mathe-Block "erniedrige Variable"

Thorlogy, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
 - Es gibt bereits folgenden Mathe-Block "erhöhe <Variable> um 1" für die Nutzerinnen könnte das Pendant "erniedrige <Variable> um 1" hilfreich sein

**Describe the solution you'd like**
 - Bereitstellung des Blocks "erniedrige <Variable> um 1" in der Kategorie Mathematik

rbudde, 2023-03-09:
I was never happy with the GERMAN name of the block ("change", "erhöhe"). But:

you can write "change/erhöhe" <var> "um/by" -1 . This decrements the value. In english this reads fine, in german it is not very intuitive.

Thorlogy, 2023-03-09:
I think - "ändere" "um" makes the most sense - in my opinion.  therefore no new block is needed.