OpenRoberta/openroberta-lab #1466: Second touch sensor within SIM

Thorlogy, 2023-03-13:
**Is your feature request related to a problem? Please describe.**
- In the simulation, it would be desirable to have a second touch sensor and align it to the rear. 
 
**Describe the solution you'd like**
If a second touch sensor is selected in RobotConfig, it should appear at the back of the simulated robot.

rbudde, 2023-03-13:
1. Currently the two touch sensor are connected to the front. They deliver always the same values .
2. there was a proposal of community members to allow a user-defined placement of sensors on the sim robot. This would increase the usability a lot and cover this proposal. Maybe the effort to add this is too high at the moment.

Thorlogy, 2023-03-13:
1. Does that make sense?
2. Of course, it would be great to let the user position the sensors. For me, the option of attaching the second sensor to the back of the robot would also be good as an intermediate step.

bjost2s, 2023-03-14:
@Thorlogy, what is your proposal for the third and fourth touch sensor?