OpenRoberta/openroberta-lab #1471: Calliope mini: Bluetooth password question

joernalraun, 2023-03-16:
When flashing (with iOS) to the Calliope mini the second time a passphrase is asked for.
Before the update several times flashing was possible without the need of a passphrase/key.

You can find an overview to the expected values here:
https://github.com/microsoft/pxt-calliope/blob/b0af4b54b2a8418fd7c9b364ece3fe2aeb70d5c1/libs/core/pxt.json#L114

This are the values for 16KB

"bluetooth": {
                    "enabled": 1,
                    "pairing_mode": 1,
                    "partial_flashing": 0,
                    "dfu_service": 1,
                    "event_service": 0,
                    "device_info_service": 1,
                    "private_addressing": 0,
                    "open": 1,
                    "security_level": null,
                    "whitelist": 0,
                    "advertising_timeout": 0,
                    "tx_power": 6
                },
                "stack_size": 1280,
                "gatt_table_size": "0x300",
                "panic_on_heap_full": 0,
                "debug": 0,
                "heap_debug": 0,
                "sram_end": "0x20004000",
                "RAM_SIZE": "\"16K\""
            }

RaghuvirShirodkar, 2023-03-17:
Hello Jørn!

It seems the issue was with this definition:
https://github.com/OpenRoberta/microbit-dal/blob/98c7be7a70669bac88bab7648ce3389ccaaf6e84/inc/core/MicroBitConfig.h#L257

As standard, `MICROBIT_BLE_OPEN` was set to 0, due to which the following definitions were not set correctly:
https://github.com/OpenRoberta/microbit-dal/blob/98c7be7a70669bac88bab7648ce3389ccaaf6e84/inc/core/MicroBitConfig.h#L261

This results in the setting of the default passkey authentication for `MICROBIT_BLE_SECURITY_LEVEL`:
https://github.com/OpenRoberta/microbit-dal/blob/98c7be7a70669bac88bab7648ce3389ccaaf6e84/inc/core/MicroBitConfig.h#L269

@bjost2s @rbudde:
I've updated the necessary definitions and tested them locally on my Calliope Mini. A new release for `ora-cc-rsc` should mitigate this issue.

joernalraun, 2023-03-17:
Yes, having the code once could have set it to the whitelist and therefore the code is not needed. However, the code should not be needed ever. When testing one should remember to always try it twice as the services are set at transfer and will be in affect the next time you are transmitting a file. From my tests Passcode appears only in Open Roberta Lab (when the second file was flashed). I could not reproduce it in MakeCode.

rbudde, 2023-03-23:
**the following works:**

- put the default firmware for BT onto the Calliope
- on Calliope: enter BT coupling by pressing A+B+reset
- on Android Calliope app: select Calliope, open private link (connects to the test server), write program, upload, program runs
- change program, upload

no pairing code is needed 

**the following is strange:**

- put the default firmware **without** BT onto the Calliope
- on Calliope: enter BT coupling by pressing A+B+reset (why does this work?)
- on Android Calliope app: select Calliope, open private link (connects to the test server), write program, upload (why does this work?), program runs 
- change program, upload, connection fails: Calliope is not detected
- put Calliope into BT pairing mode
- on Android Calliope app: select Calliope, open private link (connects to the test server), write program, upload
- NOW the question for the 6 digit pairing mode appears on the Calliope

Likely problems:
- the Caliope factory default is a BT version
- the non BT plugin/Calliope firmware continues to have the problem with the pairing code

joernalraun, 2023-03-23:
Using the test server on iPad (iOS16.3.1) works for me like expected. I tried several programs in different ways – and  transmission worked correctly. However, there is still a problem when switching editors or using the default demo-program, but this might be able to be fixed by updating the app.

closed 2023-03-24 by bjost2s