OpenRoberta/openroberta-lab #1475: Zooming in and out of the simulation fensor

Thorlogy, 2023-03-22:
**Describe the bug**
 - When you try to enlarge or reduce the right window in which the simulator is located, unpredictable behaviour occurs. The window does move. But even if you no longer want to move it, it still follows the movement of the mouse (to the left or right). Thus the window enlarges or shrinks unintentionally. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'OR Sim'
2. Click on 'SIM'
3. Move/re-size Simulation window 
4. See error

**Expected behavior**
 - Enlarge or reduce the window as long as the mouse is pressed. Releasing the "mouse" ends the re-size. 

**Device information**
 - Type: MacBook Pro
 - OS: Mac Ventura 13.2.1
 - Browser: Chrome Version 111.0.5563.64

rbudde, 2023-04-12:
Not reproducable on windows system. MAC-related problem?

The problem, that the windows follows the mouse movement (independant from zooming), is well-known, but seldom and hard to reproduce.

PascalOchel, 2023-05-03:
Could not reproduce on MacBook Pro macOS 13.3.1 (a) with latest Chrome, Firefox, Safari browsers.

Thorlogy, 2023-05-03:
Did you use the touchpad?

bjost2s, 2023-08-22:
seems to be fixed. @Thorlogy and @rbudde , please reopen if the problem still exists. Make sure you have deleted the cache with the new version.

closed 2023-08-22 by bjost2s