OpenRoberta/openroberta-lab #1476: Copying obstacles in the simulator

Thorlogy, 2023-03-22:
**Is your feature request related to a problem? Please describe.**
 - You click on an obstacle in the simulator and do copy and paste. The same object then appears again in the simulation environment. 

**Describe the solution you'd like**
 - Click on existing obstacle - press command C + V 
 - The copied obstacle is now available within the simulation environment

bjost2s, 2023-03-22:
Please check it again, copied obstacles (and color areas) apear somewhere in the environment. Multiple CTRL V leads to a stack of newly created obstacles (or color areas)
[image]

Thorlogy, 2023-03-22:
Thanks. 
Maybe it is a shortcut error in my keyboard/Mac setting. Instead of the command "command + C/V" - "control c/v" works for copy/paste in the SIM..... 
However, the shortcut "command + C/V" works on other pages/environments as usual... 
Please can you check this?

PascalOchel, 2023-03-23:
I can follow the problem. On the Mac, the convention is to use the ctrl-c and ctrl-v key combinations as well as other standard actions with the command key instead of the ctrl key. The (on the Mac so-called) command key resides on all keyboards and is also called windows or super or meta key among other operating systems.

In the Open Roberta Lab, the ctrl key is essential for copy and paste functionality. Contrary to the convention, the ctrl key must also be used on the Mac. This will probably be seen as a mistake by almost all Mac users.

I think like Thorsten that we should find a solution for this. We could certainly make the key combination dependent on the browser's user agent:

> **If** the strings _Macintosh_ or _Mac OS X_ are part of the user agent
**then** use command + c and command + v
**else** use ctrl + c and ctrl + v

PascalOchel, 2023-03-28:
Beyond Cmd+C and Cmd+V (which are easy to fix), I have found some other keystrokes that do not work as expected:

- Cmd+R in the lab "run on brick", in the text editor "align right": This is a standard keystroke of many browsers for "reload page". Safari reloads the page, Chrome and Firefox perform "run on brick" and "align right".
- Cmd+M for "my programs": In macOS this is a standard keystroke for "minimize window". Safari minimizes its window, Chrome and Firefox perform "my programs".

bjost2s, 2023-04-06:
@Thorlogy , could you please test the solution with test.open-roberta.org and comment here the result?

Thorlogy, 2023-04-06:
works fine on test

rbudde, 2023-04-12:
move to next relaease to be reviewed (again)

closed 2023-05-08 by bjost2s