OpenRoberta/openroberta-lab #1481: block "random integer" cause program abort

AHberta, 2023-03-25:
**Describe the bug**
 - When using random values via block "random integer from ... to ..." the hub aborts the running program with the RGB LED flashing red or bad smiley!

**To Reproduce**
Steps to reproduce the behavior:
1. Use block  'random integer from ... to ...' to show on LED matrix
2. Download programm
3. See error
4. Optional: Start programm - if you want to see the error once again

**Screenshots**
- Screenshot_show-text
[image]

- Screenshot_steer-random
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser Brave [Version 1.49.128 Chromium: 111.0.5563.110 (Offizieller Build) (64-Bit)]

sachinnandan, 2023-03-27:
I want to Solve this issue... Can I get this ?? @AHberta

bjost2s, 2023-03-28:
@sachinnandan , it seems that this is an error for our spike plugin, feel free to check the code generation! It will be hard to test without a Spike Prime / Robot Inventor set

closed 2023-05-08 by bjost2s