OpenRoberta/openroberta-lab #1485: IPv6 address for lab.open-roberta.org

chrschng, 2023-04-05:
This feature request is not about the lab code itself, but about the serving infrastructure. The Roberta lab at lab.open-roberta.org is currently only accessible via IPv4:

```
$ host -t aaaa lab.open-roberta.org
lab.open-roberta.org has no AAAA record
```

This means that the lab cannot be used from an IPv6-only network. This affects our Roberta trainings at Google where we operate a pure IPv6 Wifi network. 

Could you expose lab.open-roberta.org be exposed via IPv6 in addition to IPv4?

bjost2s, 2023-04-06:
Thank you for this question, I will ask our IT Services, afaik we do not have any website available with IPv6 so far.

bjost2s, 2023-04-17:
@chrschng , unfortunately there is no plan yet in our Institute to support IPv6. So as long as we host our server here in Fraunhofer, we can not do anything, but maybe we will switch, one day ....