OpenRoberta/openroberta-lab #1487: Variable Picture - Calliope min

Thorlogy, 2023-04-14:
**Describe the bug**
 - Assigning the values for the variable "hase" in the loop for further processing works only once. 
 - 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://lab.open-roberta.org --> Calliope mini
2. Code the example program
3. Start the progam and process it more then two times
4. See error

**Expected behavior**
 - Assigning the variable "hase" should work all the time. No matter how many times the loop is run..

**Screenshots**
[image]

**Device information**
 - Type Laptop
 - OS macbook
 - Browser Chorme

Thorlogy, 2023-04-14:
Indeed its SIM-related! 

Program as .txt file ;-) 
[Debug_Hase_Prog.txt](https://github.com/OpenRoberta/openroberta-lab/files/11233661/Debug_Hase_Prog.txt)

bjost2s, 2023-04-16:
looks like with the third iteration an variable "k0" is added, which has no value, so the error occurs ...

bjost2s, 2023-04-18:
While working on this problem, it turned out that we have a similar problem with the mbot image simulation. But here the hardware behaves like the simulation so the question is whether we should change both (hardware and simulation for mbot) to make it similar to Calliope?

closed 2023-05-08 by bjost2s