OpenRoberta/openroberta-lab #1495: Show Passwort

Thorlogy, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
 - Now and then it happens that you mistype your password several times. It would be helpful if you could display the password you entered. Especially if a new password has to be entered twice. 

**Describe the solution you'd like**
 - An additional button, e.g. an eye, is currently common to display one's password. 
 
[image]

perminder-17, 2023-04-28:
//assign

perminder-17, 2023-04-28:
I will add an eye from html and design it from css, then i will add an eventlistner from javascript that when i click it will show password