OpenRoberta/openroberta-lab #1506: Simulated EV3 display did not work

Thorlogy, 2023-05-12:
**Describe the bug**
Since the latest release, no texts / characters etc. are shown in the display of the simulated EV3.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'OR Lab'
2. Click on 'Open Roberta SIM or Open Roberta xNN or EV3 lejos or ev3dev, c4ev3'
3. Create a programme that displays any text in EV3.
4. See error

**Expected behavior**
 - Text / characters... will be displayed.

**Device information**
 - Type laptop
 - OS:mac
 - Browser chrome

rbudde, 2023-05-14:
deployed as hotfix 14.5.2023. Becomes active 15.5.2023 3:00 am

YogoGit, 2023-08-24:
As of 8/24/2023, this no longer works.

[image]

closed 2024-02-13 by bjost2s