OpenRoberta/openroberta-lab #1507: Strange behaviour SIM - changing Motor blocks within Robotconfiguration

Thorlogy, 2023-05-15:
**Describe the bug**
 - Changing the Motor blocks in the Robotconfiguration from B->A and C->B  
 - If only motor B is programmed, the robot moves forward. Instead of driving a curve.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'OR Lab'
2. Click on 'OR SIM'
3. Configurate the Robot and code the program shown in the screenshot
4. See error

**Expected behavior**
 - The robot drives a curve

**Screenshots**
[image]

[image]

**Device information**
 - Type laptop
 - OS: mac
 - Browser chrome

rbudde, 2023-05-15:
The reason is probably, that the simulation does not take into account, that a motor is removed from a port (say B). Even, if a new motor is connected to B. Consequently both motors start to work. This error happens if the first motor is completely REMOVED, too.

There seem to be more errors: remove the B motor, then the C motor, add a new B motor. Then the block "motor on for duration" doesn't stop anymore (ignores the duration expression)

bjost2s, 2024-07-02:
same as [#1657](https://github.com/OpenRoberta/openroberta-lab/issues/1657)

closed 2024-07-02 by bjost2s