OpenRoberta/openroberta-lab #1516: Add NN support for all robots

RaghuvirShirodkar, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
Neural network is already implemented for xNN plugin in the lab. Next step would be to introduce NN support for other robots as well.

In essence, the following needs to be carried out: 
- [ ] Add NN support as a first step (NN tab, code generation for the real robot)
- [ ] Add mutli simulation support for NN

This ticket could serve as a tracker, and the individual tasks could be their own sub issues if needed.

**Additional context**
Some robots could pose potential problems with code generation (syntax errors, for instance). This needs to be documented and possible solutions considered before moving ahead.

bjost2s, 2023-08-22:
@RaghuvirShirodkar, @rbudde please check the boxes before we close this ticket

rbudde, 2023-08-23:
solved

closed 2023-08-23 by rbudde