OpenRoberta/openroberta-lab #1519: Problem to download and run in EV3 c4ev3

dmarcelo2, 2023-06-01:
I use Open Roberta Lab online and now I'm trying to download the programming into my EV3 (c4ev3) brick. I'm using the firmware 1.10E, but after copying the .UF2 file in the EV3 it shows a Error.
I did the same test using makecode.com and the .UF2 file created works correct.
Someone can help me to undrstant what is the problem with the .UF2 file created by Open Roberta Lab ?
I'm attaching the .UF2 files created in Open Roberta and also in makecode.
[UF2 files.zip](https://github.com/OpenRoberta/openroberta-lab/files/11629708/UF2.files.zip)

bjost2s, 2023-06-02:
Could you please upload the exported program here? (change the .xml to e.g. .txt before uploading)
Could you please make a foto of the error message and upload it here?
It is quite hard for us to  reproduce this behaviour. On windows 10 everything works fine as expected.

bjost2s, 2023-06-05:
@dmarcelo2  can you please send us the (exported) program which has caused the error? It would be very helpful for us to reproduce the problem

dmarcelo2, 2023-06-09:
Hi,

To try understand what is happening I did the following tests:

1) Program "TesteAlpha" ( created in a computer with windows 10):
- The program run in the EV3 brick. The program has a "repeat
indefinetelly" block and I was supposed to stop the robot by pressing the
button of the brick, but something happens with the "stop button" of the
brick when using the .UF2 file and it was necessary to force the restart of
the brick. After restarting, the program named "TesteAlpha" is no more
showed in the EV3.
Link to video: https://photos.app.goo.gl/fVzBNNjKjAFVxZhf7
Archives (.UF2 and .XML attached to the email)

2) Program "TesteAlpha3"
- This programa don't have the "repeat indefinetelly" block and works
coorect in the brick.
  Archives (.UF2 and .XML attached to the email)

3) Program "LetraM"
- This program don't work. After restarting, other programs still works. I
did a film showing what happens. There is nothing so diferent between this
program and the "TesteAlpha3" that worked correct. I can't understand.
Link to video: https://photos.app.goo.gl/hZjmYeXfohA6kuZq9
Archives (.UF2 and .XML attached to the email)

Thanks if you can help.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Não
contém vírus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bjost2s, 2024-07-23:
When updating our servers from  Ubuntu 18.04 to 20.04 and 22.04 it turned out, that the  installation of the C4EV3 compiler results in something weird, so that the output of the compiler is different and not executable anymore. Unfortunatly the C4EV3 compiler is old and not supported anymore, so there is no workaround. 
So we decided to set the C4EV3 plugin to "deprecated". For a short time, people may use https://c4ev3.open-roberta.org to program with C4EV3 and getting a correctly compiled .uf2 file.

closed 2024-07-25 by bjost2s

timarnoldev, 2025-02-03:
Any guesses what might be the cause?
Maybe it's possible to look into the source code.
Since c4ev3 was developed in a university in my vicinity I can ask for a contact.