OpenRoberta/openroberta-lab #1525: Moving the position of variables

Thorlogy, 2023-06-12:
**Is your feature request related to a problem? Please describe.**
-Currently, the position of an initialised variable is fixed. It would be nice to be able to change the position of variables within the start block. This would bring more clarity. 

**Describe the solution you'd like**
- Moving the variables could be done with a position number or by drag and drop.

AyaMoussa, 2024-12-02:
Solved in blockly. Pull request can be found in [Pull reauests](https://github.com/OpenRoberta/blockly/pull/87)

bjost2s, 2024-12-15:
Please test this fix in detail, there are a lot of changes that could (hopefully not) break the xml

rbudde, 2024-12-16:
I tested a lot. Everything looks fine except the following error:
- create a function with param p1 and p2
- use this in some function calls
- the generated target code is fine (first image)
- move p2 to the first position
- regenerate the target code (this includes a server round-trip)
- the sequence of parameters in blockly is updated correctly
- in the code generated the sequence is wrong (second image).
I attach two images
[image]
[image]

rbudde, 2024-12-19:
works as expected

closed 2024-12-19 by rbudde