OpenRoberta/openroberta-lab #1532: Nao robot not connecting

rbtnx, 2023-07-13:
For approximately 3 weeks now all of our Nao robots are not connecting to OpenRoberta Lab. I didn't change anything and only had a look at main.py on the robot to see what's going on. The server responds with 404:

    [DEBUG] - 2023-07-13 16:56:37.475379 - JSON decoding error (robot was not registered within timeout), reconnecting in 10 seconds...
    
    /usr/lib/python2.7/site-packages/urllib3/connectionpool.py:857: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
      InsecureRequestWarning)

    [DEBUG] response-dump ( --> server_response.content )
    <html>

    <head>
    <meta http-equiv="Content-Type" content="text/html;charset=utf-8"/>
    <title>Error 404 Not Found</title>
    </head>
    <body><h2>HTTP ERROR 404</h2>
    <p>Problem accessing /pushcmd. Reason:
    <pre>    Not Found</pre></p><hr><a href="http://eclipse.org/jetty">Powered by Jetty:// 9.4.15.v20190215</a><hr/>
    
    </body>
    </html>

The request that Nao sends looks like this:

```javascript
{'files': [], 'cookies': None, 'url': 'https://lab.open-roberta.org/pushcmd', 'hooks': {'response': []}, 'auth': None, 'headers': {'Content-Type': 'application/json'}, 'json': None, 'params': {}, 'data': '{"menuversion": "1.0.5", "token": "TOKEN", "macaddr": "MAC ADDR", "nepoexitvalue": "0", "battery": 99, "firmwarename": "Nao", "cmd": "register", "robot": "nao", "firmwareversion": "2-8", "brickname": "nao"}', 'method': 'POST'}
```

The main.py script says the endpoint should be /rest/pushcmd so I manually also tried to send the request to https://lab.open-roberta.org/rest/pushcmd, this is the answer:
```json
{"rc":"error","cause":"SERVER_ERROR","server.version":"5.2.2","message":"SERVER_ERROR","_version":"1","initToken":"error","server.time":1689263905130}
```

**Robot information**
 - Nao v6
 - naoqi version: 2.8
 - Open Roberta Lab Connector version: 1.0.5 (latest)

bjost2s, 2023-07-18:
This is a bug and we know how to fix it, but unfortunately we do not have a "working" NAO by hand to test it. Have you installed the connector via Choregraphe? If so, I could prepare a fix and you might test it (for us)? If you have installed the connector via the aldebaran store it will be difficult, I think.

rbtnx, 2023-07-19:
I did not install the connector, so I don't know how it was installed. I always start the app via Choregraphe and I found the python scripts on the nao itself in

    ~/.local/share/PackageManager/apps/v6_openroberta_connector/

I can manually patch the files and test it or reinstall the connector via Choregraphe, though.

bjost2s, 2023-07-27:
this issue seems to be related to #1026, so the fix for https://github.com/OpenRoberta/robertalab-naoclient will be to add "rest" to the URL

bjost2s, 2023-07-27:
It seems that this [commit](https://github.com/OpenRoberta/openroberta-lab/commit/e9845d8b36e31ee98a0fcf742bc54637ce5f5922) removed the deprecated REST API without prefix. Unfortunately this was still used in the [naoclient](https://github.com/OpenRoberta/robertalab-naoclient).

bjost2s, 2023-07-27:
Please checkout the latest release https://github.com/OpenRoberta/robertalab-naoclient/releases/tag/v1.0.8. It should work now: connecting to the lab and downloading user programs. Please let me know if this fixes your problem.

rbtnx, 2023-08-04:
Thank you very much, worked like a charm!