OpenRoberta/openroberta-lab #1542: Migrate Spike Prime Robot to Pybricks Firmware

heini208, 2023-08-02:
Currently, the Lab uses outdated firmware for the spike prime plugin. Due to difficult communication with Lego, the open-source firmware [pybricks](https://pybricks.com/) could be used as the firmware from now on.
 
This includes:
     - code generation for the new API
     - new connection method, pybricks currently only supports Bluetooth, but USB is possible in cooperation with the developers.   

- [ ] Code generation new API
- [ ] Bluetooth Connection
- [ ] USB Connection
- [ ] Integration Tests
- [ ] Adjust documentation for firmware

DrTom, 2023-10-23:
Can someone give any indication if this will be implemented?

bjost2s, 2023-10-30:
it is very likely that it will be implemented, soon

closed 2024-06-27 by bjost2s