OpenRoberta/openroberta-lab #1551: Robotino config fails validation after import

heini208, 2023-08-25:
**Describe the bug**
 - The ports "BN" and "BU" of the optical sensors are recognized as already taken after importing 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to robotino
2. export any program
3. import it
4. try starting the program or opening the simulation
5. see error

**Expected behavior**
 - The ports BN and BU should not be checked whether or not they are taken

**Additional context**
 - The AbstractRobotinoValidatorAndCollectorVisitor already includes "BN" and "BU" as NON_BLOCKING_PROPERTIES but the ports change after imports from BN/BU to +/-
 - This means there is likely a difference between the set default configuration and the blockly implementation of the configuration component.
 - changing the NON_BLOCKING_PROPERTIES to this:
`    private static final List<String> NON_BLOCKING_PROPERTIES = Collections.unmodifiableList(Arrays.asList("BN", "BU", "M1", "M2", "M3", "+" , "-"));`
fixes the issue but it should rather be fixed that the ports do not change after import

heini208, 2023-08-28:
This seems to be a general issue with fixed ports. After exporting any fixed port they change from something like ['VCC', '5V'], to ['5V', '5V'] in the XML. 

This is only caught by the robotino because of an automated checker for every port name, which needs to exclude ports if they are not supposed to be checked. In this case, the ports BN/BU are excluded, but after importing, the name is changed to +/-.

A workaround would be to add +/- to the excluded ports. This is a very simple fix that is done in [this](https://github.com/OpenRoberta/openroberta-lab/pull/1556) PR.

heini208, 2023-09-10:
The issue has been resolved in [this](https://github.com/OpenRoberta/blockly/commit/2642d97b45f797a6d2a86051f113367cecb23986) Blockly commit and the Workaround is not longer needed. Instead only the tests of the robotino should be adjusted to include the fixed ports. This way the Integration tests would catch an error like this one.

bjost2s, 2023-11-14:
tested

closed 2023-11-14 by bjost2s