OpenRoberta/openroberta-lab #1579: MBot2 zeigt 'NO_PORT' im Programmierblock

rbudde, 2023-11-14:
- select MBot2
- see in the configuration, that there are two encode motors named R and L
- switch to the programming tab and select the drive block from the advanced tool box

- the block shows NO_PORT

The Block has to show a dropdown with the values R and L, R pre-seelcted

rbudde, 2023-11-15:
see the attached image: one of the two blocks is affected only
[image]

heini208, 2023-11-15:
Caused By the added && !'joycar' [here](https://github.com/OpenRoberta/blockly/blob/e976d624050459a84525681a3ca379086a303f0b/blocks/robActions.js#L208C50-L208C50) 

!this.workspace.device === 'joycar' is always false which causes encoders to not work here anymore.

heini208, 2023-11-15:
Resolved in [this](https://github.com/OpenRoberta/openroberta-lab/commit/5dbb123ef2ce2ade218f4f6e74fa2f7c734526dd) Commit adding [this](https://github.com/OpenRoberta/blockly/commit/53281c54c17927d8c6c9745f63ff6510bb028eda) Blockly Commit.

closed 2023-11-17 by bjost2s