OpenRoberta/openroberta-lab #1580: Simulation lacks precision when sensor values are read from the environment

rbudde, 2023-11-16:
See the attached pictures of a background and a program
[image]
[image]
Optionally open the sensor view.
The background colour seen by the sensor is displayed in the small robot. The update of the view is MUCH TOO LATE. The robot has left a colour area for a long time and the colour stays visible. This was not the case in earlier version of the simluation.
If the colour is used in a program, the same is true: the information is too late.

Remedy: the time between (the background delivers new info) and (the simulation makes the info available) should be much smaller.

Thorlogy, 2023-11-20:
Reproduction of the error: 

1. Open Roberta SIM,
2. Click on SIM
3. Change the scene (its not a must)
4. Click on "Add a color area"
5. Draw the color area
6. Close the SIM
7. Open  the SIM 
8. See the error

rbudde, 2023-11-20:
yes, the **close and open** of the sim trigger the error.

It seems to me, that the color area is painted at wrong coordinates (?)

PamoCurtis, 2024-02-13:
solved

closed 2024-02-13 by PamoCurtis