OpenRoberta/openroberta-lab #1581: EV3 programms do not run

othreed, 2023-11-19:
**Describe the bug**
 - NEPO programs downloaded on EV3  (firmware v1.10E) C4ev3 via USB do not run. The programs immediately exit.
 

**To Reproduce**
Steps to reproduce the behavior:
1. Program a Nepo program for EV3
2. Connect the EV3 via USB and download the program to it via USB (C4ev3)
3. Start the program on the EV3
4. Program starts and immediately exits.

**Device information**
 - Type [Desktop]
 - OS: Windows 10, OSX 14.11
 - Browser Chrome, Firefox, Safari

bkochsickm, 2023-11-22:
Additionally, the EV3 often loses connection.

bkochsickm, 2023-11-23:
Kopiert aus einem anderen Ticket (dort als Kommentar): 

[charlie-chaperon](https://github.com/charlie-chaperon) commented [last week](https://github.com/OpenRoberta/openroberta-lab/issues/1549#issuecomment-1816623908)
This problem appeared to me just recently (15th november 23) during my weekly robotics workshop. I investigated by using several platforms and I think I could locate the problem.

Tested systems

several versions of Win 10 Enterprise and Pro - 22H2 - Builds 19045.3636 & 19045.3208 & 19045.3570
Ubuntu Linux 22.04.3 LTS
Debian 12
tested browsers
Firefox 119.0 & 115.3.1.esr
Chrome 119.0.6045.124 and other recent versions
recent Brave browser

If you need more precise information about the systems, let me know. I stopped writing down details when I found the potential cause.

What does work, what does not?
Open Roberta lab 5.2.6 + "Save as" = works on none of the systems tested (Win10)
Open Roberta Lab 5.2.6 + upload-c4ev3 script (Linux) = does not work, error message during script
Lego Mindstorms app (1.5.2 & 1.5.0) on Win10 = works without problems
Open Roberta lab 5.2.6 + ev3dev uplaod via Wifi / internet = works without problems
uploading old (april 2023) uf2 programs = works on all linux (via script) and Win10 systems (via send to and direct copy to ev3)

last known working setup = 8th november 23 on Win10
I started a weekly robotics workshop where we used Open Roberta Lab. It worked without the issue until Wednesday 8th. On my workshop this wednesday (15th), none of the laptops and/or EV3 bricks did work. When I checked the issue at home, I realized that my other EV3 bricks which had no problems before showed the same problem [see tested systems].

symptoms
The usual upload process (save the program file from browser to brick) does not work. I don't get the beeping sound to confirm the upload. Although there is a "NEPOprog" visible after download (also on systems that did not have any present before), it seems to be transferred incorrectly. Instead of being placed inside a seperate folder (like NEPOprog/NEPOprog.uf2) it seems to be located directly at BrkProg_SAVE , next to the demo program, when I look for it via the EV3 screen. When I try to start the program (NEPOprog) the Mindstorms screen appears, but immidiately stops, so i can't even read the output.

When using the upload-c4ev3 script in linux, i get the following output:

linux console:

:~/Downloads$ upload-c4ev3 NEPOprog.uf2

Extracting UF2 file
UF2 file successfully extracted

Uploading program files ...
(ev3duder output follow)
USB connection established.
upload was successful.
USB connection established.
upload was successful.
USB connection established.
upload was successful.
USB connection established.
upload was successful.
(end of ev3duder output)
(ev3duder output follow)
USB connection established.
../prjs/BrkProg_SAVE/NEPOprog.elf: /lib/libc.so.6: version GLIBC_2.33' not found (required by ../prjs/BrkProg_SAVE/NEPOprog.elf) ../prjs/BrkProg_SAVE/NEPOprog.elf: /lib/libc.so.6: version GLIBC_2.34' not found (required by ../prjs/BrkProg_SAVE/NEPOprog.elf)
../prjs/BrkProg_SAVE/NEPOprog.elf: /lib/libc.so.6: version GLIBC_2.32' not found (required by ../prjs/BrkProg_SAVE/NEPOprog.elf) exec` status unknown.
(end of ev3duder output)
Program files uploaded

Done!
Wait for the robot to beep and then start the program

EV3 screen (while uploading):

executing shell cmd

This message is shown ~ half a second

suggested problem
It looks like a glibc related issue to me. My linux systems use glibc 2.36-9+deb12u3 or 2.35-0ubuntu3.4 respectively, but I assume my laptop's version is not the problem. Since the problem appeared just recently (Open Roberta Lab was released novemeber 14th) and I had no problems before, I suppose this happened due to recent changes.

workaround
Uploading older uf2 programs (april 2023) that I had still around worked just fine with the same laptops and without the error message. Using Open Roberta Lab 5.2.5 might just work.

to do

 check if older version are affected

bjost2s, 2023-11-24:
It seems that the C4EV3 compiler no longer works on a current Ubuntu version (we recently upgraded to 22.04). Since this project (the C4EV3 compiler) was a student project, we can't hope for updates but are currently working on a workaround here. But that will take some time. As soon as I have new information, I will let you know here.

charlie-chaperon, 2023-12-20:
I tested the uplaod with Lab version 5.2.8 and Windows 10 as well as Linux work again.
Thanks a lot!

bjost2s, 2024-01-02:
solved by redeployment

closed 2024-01-02 by bjost2s