OpenRoberta/openroberta-lab #1587: Spike: pink color-block double available

AHberta, 2023-12-02:
**Describe the bug**
 - Pink color block is duplicated in the Colors category

**To Reproduce**
Steps to reproduce the behavior:
1. Click on 'Colours'
2. See the color of the 2nd and 5th color block are identical

**Screenshots**
[image]

**Device information**
 - Laptop/Desktop/etc.
 - Windows/etc.
 - Chrome/etc.

NiHoffmann, 2024-03-06:
pybricks has an additional color distinction please also add the color brown to blockly for color sensor and picker

bjost2s, 2024-03-07:
@NiHoffmann , this issue is only for the toolbox in this repo. So for adding a new color, please add a new issue.

rbudde, 2024-03-07:
tested.

closed 2024-03-07 by rbudde