OpenRoberta/openroberta-lab #1590: xNN - Input Trainings data on Ipad

Thorlogy, 2023-12-04:
**Describe the bug**
The pop-up window for entering training data does not appear on iPads. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use a ipad
2. Click on 'Open Roberta xNN'
3. Click on "Neuronal Network learn"
4. Click on "Open a pop up window - input/output test/training data"
5. See error

**Expected behavior**
Pop up window appears. 

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type iPad
 - OS: 
 - Browser: Safari, Chrome?

bjost2s, 2024-01-02:
@Thorlogy , please add more information for Ipad version and browser version. We could not reproduce the error

rbudde, 2024-01-08:
@Thorlogy please try it again. We could not reproduce the error.

closed 2024-05-07 by bjost2s