OpenRoberta/openroberta-lab #1592: xNN - scale browser window

Thorlogy, 2023-12-04:
**Describe the bug**
The "training loss" display window should not change when the browser window is minimised from bottom to top. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Open Roberta xNN'
2. scale browser window (horizontal)
3. See error

**Expected behavior**
training loss display should be the same

**Screenshots**
[image]

**Device information**
 - Type Laptop
 - OS: mac
 - Browser chrome

TomGruene, 2024-02-13:
not solved

bjost2s, 2024-03-07:
should be fixed with together with c4136b6

closed 2024-03-07 by rbudde