OpenRoberta/openroberta-lab #1596: Sim: No deletion of the robot draw track with "reset the positions ..."

AHberta, 2023-12-08:
**Describe the bug**
 - In all simulations that support the "Draw" feature, the drawn robot track is retained after activating the "Resets the positions ..." button. Deletion is not possible in such simulations.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'SIM'
2. Click on 'Enable/Disable robot draw trail'
3. Start program for example drive forward speed %30 distance cm 20
4. See 20 cm drawn robot track 
5. Click on 'Resets the positions ...'
6. See the position reset and still the drawn track

**Screenshots**
[image]

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - Add any other context about the problem here.

rbudde, 2023-12-08:
it is a bug or it is a feature? I use line painting sometimes to see which area is covered by a running robot. In this case I prefer, that the lines is not erased.

Design alternatives:
- erase by resetting location
- erase when the 'draw line' "checkbox" (it is a kind of stateful checkbox) is reset
- leave it as a feature

if you close and reopen the sim, the lines stays visible. At least this is an error.

AHberta, 2023-12-09:
Bug or feature? The behaviour of release 5.2.8 (lab.open-roberta.org) and 5.2.2 (dev8.open-roberta.org) are different:
 
5.2.2
draw checkbox enabled + reset position click = **drawn track is erased**
draw checkbox disabled + reset position click = **drawn track is erased**
Close and re-open the SIM = drawn track is still visible
Change background = drawn track is erased
Open Debug Mode = drawn track is still visible + reset robot position + draw checkbox disabled (even if it was enabled)

5.2.8
draw checkbox enabled + reset position click = drawn track is unerased
Draw checkbox disabled + reset position click = drawn track is unaresed
Close and reopen the SIM = drawn track is still visible
Change background = drawn track is erased 
Open Debug Mode = drawn track is still visible + reset robot position + draw checkbox disabled (even if it was enabled)

If not erase the drawn track when resetting the robot position via [image] is a new feature, then the info text is irritating:
[image] 
[image]

bjost2s, 2024-01-25:
@AHberta, thank you so much for clarification. I have fixed this problem and it is now working like it was proposed in 5.2.2 and in the tooltip!

NiHoffmann, 2024-02-13:
solved

closed 2024-02-13 by NiHoffmann