OpenRoberta/openroberta-lab #1599: Blocks placement crash

Sinclair732, 2023-12-27:
When loading/import the attached program (Nepo4Arduino Uno) and try to add more blocks (or copy), the blocks placement is crashed and the code view is wrong. I have equal problems too when saving the project to my programs (online) or export and try to reload it.

Steps to reproduce the behavior:
1. Import the attached Nepo4Arduino Uno project 
2. Show and refresh the source code view (everything fine)
3. Copy the last if-block of the main-loop (if ... jetztVorwaerts or statusVorwaerts)-block
4. move the copied block directly behind the original if-block (directly before the 100ms wait)
5. Show and refresh (1-2 times) the source code view (Blocks-view placement completely wrong, code view is old state)

or:

1. Import the attached Nepo4Arduino Uno project 
2. Show and refresh the source code view (everything fine)
3. Copy the last if-block of the main-loop
4. export the program without online-saving or refreshing the code-view
5. import the program shows Warning message: <Filename> ist kein gültiges NEPO Programm und kann nicht hochgeladen werden!

**Project export**
[NEPOprog3.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/13774244/NEPOprog3.xml.txt)
(remove .txt extension before import)

 **Screenshots**
[image]

**Expected behavior**
 - Save/Export and reload/import of the (larger) program should work.

**Device information**
 - Type: Desktop
 - OS: Debian Bullseye
 - Browser Firefox 115.6.0

Please ask if more information is helpful. Local server (Debian VM) with current git clone is also available, but i'm not a java expert, so debugging is not as simple.

(sry of not so good english...)

rbudde, 2024-01-08:
I reduced the program to the one attached. The error is in blockly. There is a difference between 'drag vom toolbox' and 'copy'. Load the program attached and then:
- click show source: ok
- drag 'm' from toolbox and put it into the second if, click show source: ok
- copy 'm' from the first if, move it to the second, click show source: error

[NEPOprog3-reduced.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/13859822/NEPOprog3-reduced.xml.txt)

I expected, I could reproduce this error with other plugins or with other Arduino blocks. But I couldn't.
Probably a defect with the arduino call method block

rbudde, 2024-01-08:
should be solved with the next release mid Februar.

AyaMoussa, 2024-02-13:
Not Solved. Copying blocks still makes the program crash.

closed 2024-10-25 by bjost2s