OpenRoberta/openroberta-lab #1600: mbot2 drive curve with distance doesn't drive correct distance (in some cases)

NiHoffmann, 2024-01-03:
mbot does not drive correct distance for the following cases:
1. one motor speeds is set to 0: 
try setting: 
rpm left: 0
rpm right: 30
distance: 200, 2000, 3000
drive time is below one second for all these values which is noticeable to short, driven distance doesn't come close the expected value
2. one motor is counter rotating:
rpm left: -10
rpm right: 30
distance: 200, 2000, 3000
drive time is below one second for all these values which is noticeable to short, driven distance doesn't come close the expected value

for both of these examples drive time will be below one second, note that for 2000cm the robot drive longer than for 3000cm
and about the same time for 200cm and 3000cm

Cases that work:
note that if both motor speed are defined in the same direction the robot appears to be driving the correct distance e.g. 
rpm left: 30
rpm right: 30
distance: 20cm
will drive 20cm 
same goes for :
rpm left: -30
rpm right: -30
distance: 20cm
will drive 20cm but backwards

if booth speeds or distance is set to 0 the robot wont move which is expected behaviour

when driving a curve and not just straight forward the robot appears to be driving the correct distance but this might have to be checked thoroughly again.

VinArt, 2024-01-04:
I think the issue is in getTimeToWait function, if I understand it correctly, for differential drive we'd need to compute how long would it take the center of the axis to move a given distance, so instead of computing absoluteDistance we should use distance instead.

Then for -0.9 and 1 RPM for distance of 1 m we'd get 5876.49 seconds instead of 9.5. Another thing, when resulting speed is 0 then (theoretically) an infinity should be returned, not 0. Technically it makes sense to throw an exception (e.g. invalid robot speed) also in case when RPM left is the negative of RPM right, an exception should be thrown (e.g.infinite run time requested) because in this case the robot will spin in place forever.

I think it is sin that causes this:
```
rpm L/R, distance: -0.9 1 100
time to wait: 9.560226873429032
rpm L/R, distance: -0.9 1 1000
time to wait: 10.153591611176166
rpm L/R, distance: -0.9 1 10000
time to wait: 3.6503858402500824
rpm L/R, distance: -0.9 1 100000
time to wait: 15.636709165184454
rpm L/R, distance: -0.9 1 1000000
time to wait: 17.21935670254879
rpm L/R, distance: -0.9 1 10000000
time to wait: 10.256383147792251
rpm L/R, distance: -0.9 1 100000000
time to wait: 2.41391530172498
rpm L/R, distance: -0.9 1 1000000000
time to wait: 17.396190869979414
rpm L/R, distance: -0.9 1 10000000000
```
time to wait oscillates, instead of steadily growing

when using distance instead of absoluteDistance:
```
rpm L/R, distance: -0.9 1 100
time to wait: 5876.490206469974
rpm L/R, distance: -0.9 1 1000
time to wait: 58764.90206469974
rpm L/R, distance: -0.9 1 10000
time to wait: 587649.0206469974
rpm L/R, distance: -0.9 1 100000
time to wait: 5876490.206469974
rpm L/R, distance: -0.9 1 1000000
time to wait: 58764902.06469974
rpm L/R, distance: -0.9 1 10000000
time to wait: 587649020.6469973
rpm L/R, distance: -0.9 1 100000000
time to wait: 5876490206.469974
rpm L/R, distance: -0.9 1 1000000000
time to wait: 58764902064.69974
rpm L/R, distance: -0.9 1 10000000000
time to wait: 587649020646.9973
```

TL/DR: distance needs to be used instead od absoluteDistance, inf should be returned for the case of the robot spinning in place, or an exception needs to be raised

heini208, 2024-02-13:
Fixed: Tested with different speed values.

closed 2024-02-13 by heini208