OpenRoberta/openroberta-lab #1603: Error in "neuralnetwork.ui.js" when switching to any language

bjost2s, 2024-01-06:
To reproduce this error open the lab, select any robot/system and select any other language:

[image]

It is not clear to me whether this error has further consequences, but the desired solution should avoid this error.

rbudde, 2024-01-08:
solved in commit 1e32d685dc5149399dcbd25522e73a8d4402fbaf

bjost2s, 2024-01-10:
Unfortunately https://github.com/OpenRoberta/openroberta-lab/commit/1e32d685dc5149399dcbd25522e73a8d4402fbaf does not solve the problem. The error message still appears as soon as you switch the language.

rbudde, 2024-01-11:
there seem to be a number of problems of this type :-< . Next try

TomGruene, 2024-02-13:
solved

closed 2024-02-13 by TomGruene