OpenRoberta/openroberta-lab #1604: Show Time Value OR SIM

Thorlogy, 2024-01-08:
**Describe the bug**
- No Time Value will be displayed
- 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'OR Lab'
2. Click on '.any Hardware System e.g. OR Sim'
3. Choose / code a program
4. Click on Sim
5. Click on Open Sensors data view
6. Click on Time Value
7. See error

**Expected behavior**
Time Value will be displayed

**Screenshots**
[image]

**Device information**
 - Type Laptop
 - OS: MacOS
 - Browser: Chrome

rbudde, 2024-01-08:
I tested with a program, which contains a timer (reset and read). The value is not shown on the sensor data view

bjost2s, 2024-03-05:
tested

closed 2024-03-05 by bjost2s