OpenRoberta/openroberta-lab #1608: [Calliope mini][micro:bit] Messages->Set channel to bug in some languages

fabianhugo, 2024-01-17:
**Describe the bug**
The block "Set channel to ..." is not usable in some languages, since it does not leave space to set a variable number and shows _1%_.
[image]
Correctly it would look like this: 
[image]
Checking briefly I found the following languages affected: 
- Svenska
- Italiano
- the language two below Catalá
- Polski

The "set channel to ..." block is particularly important when working with a bigger group of children.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Open Roberta Lab 
2. Use Calliope mini or micro:bit
3. Go to Extended Blocks (Click on the number 2)
4. Go to messages, use "set channel to" block
5. Change language to one of the above e.g. Italiano

**Expected behavior**
 - The block should leave space for a parameter.

bjost2s, 2024-01-19:
Thank you for this bug report!

AyaMoussa, 2024-02-13:
Solved

closed 2024-02-13 by AyaMoussa