OpenRoberta/openroberta-lab #1609: Calliope mini - get LED brightness x y reversed within SIM

Thorlogy, 2024-01-18:
**Describe the bug**
 - The x and y axes were swapped in the simulation display.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Calliope mini plugin
2. Code a program using the "get LED brightness x y" Block
3. Click on SIM and run program
4. See error

**Expected behavior**
 - The x and y axes should be displayed correctly

**Screenshots**
[image]

**Device information**
 - Type Laptop
 - OS: max
 - Browser chrome

thoschi, 2024-01-19:
Dupiclate of #1585 (but good to know I am not the only one)

AyaMoussa, 2024-02-13:
Solved.

closed 2024-02-13 by AyaMoussa