OpenRoberta/openroberta-lab #1615: URL and URL parameters needs refactoring

bjost2s, 2024-02-07:
Currently there is an old and a new way to pass parameters with the url.
Old style e.g.
- localhost:1999#overview
- ocalhost:1999#loadSystem&&ev3lejosv1
- localhost:1999#tutorial
- localhost:1999#tutorial&&bionics4education1
- localhost:1999#tutorial&&bionics4education1&&kiosk
- https://lab.open-roberta.org#forgotPassword&&_vC353v-9Pr_
- https://lab.open-roberta.org#activateAccount&&-Ik9R-7-vdWo

The new search style: domain?param1=value1&param2=value2 .... (introduced years ago) should be used from now on, at least all URL's that are generated by the server itself, others should be marked as deprecated:
- localhost:1999?activateAccount=_vC353v-LPr_
- localhost:1999?tour=overview
- localhost:1999?tour=welcome
- localhost:1999?loadSystem=ev3lejosv1
- localhost:1999?loadSystem=ev3lejosv1&tutorial
- localhost:1999?loadSystem=ev3lejosv1&tutorial=bionics4education1
- localhost:1999?loadSystem=ev3lejosv1&tutorial=bionics4education1&kiosk
- localhost:1999?loadSystem=ev3lejosv1&exampleView
- localhost:1999?loadSystem=ev3lejosv1&loadProgram=<export> ... </export>

PamoCurtis, 2024-02-13:
At some point the '#' symbol is added to the URL which leads to the creation of a faulty program link like this one: https://test.open-roberta.org/#?loadSystem=ev3lejosv1&loadProgram ...

bjost2s, 2024-02-13:
solved

closed 2024-02-13 by bjost2s