OpenRoberta/openroberta-lab #1622: Open Roberta Sim: Add colour area button flickers

AHberta, 2024-02-10:
**Describe the bug**
In the simulation, the button for adding a coloured area flickers, if the mouse pointer is on the button. The same problem also exists with the button for adding an obstacle.

**To Reproduce**
Steps to reproduce the behavior:
1. Go into Sim
2. Move the mouse to the "Add a colour area" button - button flickers
3. Move the mouse to the "Add an obstacle" button - button flickers

**Screencast**

https://github.com/OpenRoberta/openroberta-lab/assets/19265064/2506b3f6-86c0-4523-a1b7-14040064169c
Unfortunately not visible in the screencast

**Device information**
 - System: Open Roberta Sim
 -  OR: test.open-roberta.org
 - Type Laptop
 - OS: Windows
 - Browser Edge

AyaMoussa, 2024-12-03:
fixed on [feature/Issue1622](https://github.com/OpenRoberta/openroberta-lab/tree/feature/Issue1622)

AyaMoussa, 2024-12-12:
I have updated the solution to activate auto-close for the dropdowns. 
To clarify: the bug is mainly about the tooltip being associated with the icon rather than the button itself, this makes the button flickers if the tooltip and the mouse pointer overlap
I tried to show it in the video
[Screencast from 12.12.2024 17:02:39.webm](https://github.com/user-attachments/assets/f0e6dc97-1c73-42bc-9a68-402c17fa5dd1)
.

rbudde, 2024-12-16:
works as expected.

closed 2024-12-16 by rbudde