OpenRoberta/openroberta-lab #1623: Simluation - colour field

AHberta, 2024-02-10:
**Describe the bug**
In the simulation, the colour field for changing the colour of a selected coloured area or obstacle is only partially visible in the browser window.

**To Reproduce**
Steps to reproduce the behavior:
1. Go into Sim
2. Add colour area
4. Click on 'Change colour"
5. See the colour field only partially - 'Change colour' near the browser window edge

**Screenshots**
[image]

**Device information**
 - OR: test.open-roberta.org
 - Type Laptop, iPad
 - OS: Windows, iOS
 - Browser Chrome, Edge, Safari

AyaMoussa, 2024-10-10:
Solved on branch [feature/Issue1623](https://github.com/OpenRoberta/openroberta-lab/tree/feature/Issue1623)

AyaMoussa, 2024-10-22:
Tested. Issue solved.

closed 2024-10-22 by AyaMoussa