OpenRoberta/openroberta-lab #1630: wiki page for windows problems with microbit etc. is lost

rbudde, 2024-02-23:
in the prod lab, select microbit. You see the note about windows problems. Beside the fact, that the text is not completely clear, it offers a link to solve the windows problem. But trhe page doesn't exist.

This page should be made visible as soon as possible.

rbudde, 2024-02-23:
btw: when I copy the hex file directly to the calliope (as in the past, using the Calliope as storage device), the process succeeds. This was not true some weeks before. Has Microsoft solved the problem?

Other Windows users: please test it, too.

bjost2s, 2024-02-23:
Thank you very much! For the next issue in this area, please contact me directly because neither the wiki nor the notes are part of this repository. Btw, someone has deleted the page, which should never happen!!!

closed 2024-02-23 by bjost2s