OpenRoberta/openroberta-lab #1633: Edison and xNN

rbudde, 2024-02-24:
analysing our code generator, that could be used for Neural Nets on the Edison, some problems arise due to the fact, that Edison Python is different to "standard" Python:

1. in Edison Python global declaration can only be used in function definition. In Python they can be used top level, too.
2. in Edison Python in function definitions there is only one global allowed as the _first_ statement. In Python the may be many declarations, everywhere, but before the first use
3.  in Edison Python variables can be accessed before they are initalized. The program behavior is undefined.

Especially the Python standard behavior of 1+2 are used in our NN code generator. The code generator must be rewritten to follow the restrictions of Edison Python.

Example program for topic 1+2:
```
    print('start')
    global g # legal in standard Python, illegal in Edison Python
    g = '1'
    h = '2'
    
    def f1():
        print('start f1')
        global g # legal in standard Python, illegal in Edison Python
        print('g is: ' + g)
        g = '3'
        # global h
        print('h is: ' + h)
        # h = '4'  # if used, of course a global decl is needed
        print('end f1')
    
    f1()
    print('g is: ' + g)
    print('h is: ' + h) 
```

Example program for topic 3 **(this program is of minor interest for the xNN compilation)**:
```
#-------------Setup----------------
import Ed
Ed.EdisonVersion = Ed.V1
Ed.DistanceUnits = Ed.TIME
Ed.Tempo = Ed.TEMPO_MEDIUM
#--------Your code below-----------

# x = f2() # illegal in standard Python: f2 is undefined

def f2():
    global x
    return x # legal in Edison Python, in spite of the fact, that x is undefined

x = f2()  # this line is legal, but return x is illegal in standard Python, as x is undefined
```

bjost2s, 2024-02-28:
In #1516 we have decided to not support NN for edison, so it would be helpful to keep this issue up to date and / or create a wiki entry for NN support (perhaps a little longer lasting).

bjost2s, 2024-03-05:
tested

closed 2024-03-05 by bjost2s