OpenRoberta/openroberta-lab #1642: steer wont run motors backwards on NXT

Andy1978, 2024-03-05:
If you change direction of the steer block, the motors don't turn backwards when speeds are different for left and right.
The simulation shows the robot moving as expected.

robot configuration
[image]

program NEPOprog
[image]

Although the documentation for the "steer" block isn't very detailed I would expect to reverse the movement with either the dropdown field "forward/backwards", by inverting both speed setting or inverting the distance. These three work is both speeds are equal. As soon as they slightly differ (for example 29% and 30%), moving backwards no longer works and results in an endless waiting state.

NXT Firmware 1.29.2

Andy1978, 2024-03-05:
The problem is in https://github.com/OpenRoberta/ora-cc-rsc/blob/master/RobotNXT/NEPODefs.h#L120 where OnFwdRegEx is called with powerLeft and powerRight with floats and not chars. Explicitly copying to char before calling OnFwdRegEx fixes it, although I think this is a dirty hack

```
    char tmp_powerLeft = powerLeft;
    char tmp_powerRight = powerRight;
    if (absPowerLeft < absPowerRight) {
        OnFwdRegEx(portLeft, tmp_powerLeft, OUT_REGMODE_SPEED, RESET_ALL);
        OnFwdRegEx(portRight, tmp_powerRight, OUT_REGMODE_SPEED, RESET_ALL);
...
```

bjost2s, 2024-03-07:
Thank you very much, for the issue and for the solution! Do you have any timing issues with this solution? School or workshop dates, etc? If not, we plan to update the ora-cc-rsc repo at the end of this month. 
Until then, we will integrate the fix on one of our test servers so that it can be tested. I will let you know here which test server the fix is on as soon as it is possible.

Andy1978, 2024-03-22:
I made various tests with different NXT firmwares:

* Debian GNU/Linux has a package "nxt-firmware - Improved firmware for LEGO Mindstorms NXT bricks" .  In Debian 12 (bookworm), this installs V1.29.2 from this upstream https://git.ni.fr.eu.org/nxt-firmware.git/ which is based on the last available Lego sourcecode released (V1.29) together with some improvements (for example https://nxt-firmware.ni.fr.eu.org/changes/absolute_position_regulation/) and builds with gcc (original Lego firmware used the IAR compiler). Nicolas Schodet is the maintainer of this repo and it's still alive. I'll refer to this firmware using "NXT Improved firmware"
* Then there is "John Hansen Firmware", aka "NXT Enhanced Firmware" aka "NBC/NXC Firmware". The latest binary is https://bricxcc.sourceforge.net/test_releases/lms_arm_nbcnxc_132_20130303_2051.rfw but as far as I know he lost his sources for that some time ago. There is still a subversion repo https://svn.code.sf.net/p/mindboards/code/ but the latest changes are missing. I'll consider this repo and the bicxx code as abandoned since there was no change since more than 10 years and John Hansen doesn't reply to my emails. I'll refer to this firmware as "NXT Enhanced Firmware"
* The original Lego Firmware is still available as binary LEGO_MINDSTORMS_NXT_Firmware_V1.31.rfw but there is no Sourcecode available for this although V1.29 was released as OpenSource. I tired to contact the Lego customer service multiple times but they refuse to give me the sourcecode for V1.31. Here is a link from someone else trying to find the source https://bricks.stackexchange.com/questions/14785/online-link-to-the-lego-nxt-firmware-v1-31-source-code-or-iar-arm-lego-develop

So, finally back to this bug report.
* The Lego firmware binaries V1.29 and V1.31 works as expected with SteerDriveEx with and without my patch
* The "NXT Improved firmware" needs my patch up to V1.29.5 (but fixed on the float-signed branch)
* The "NXT Enhanced Firmware" needs my patch

I would understand if OpenRoberta took the position that their software must only work with the original Lego firmware but I would welcome an implementation that also runs on the free firmwares based on LEGOs latest released sourcecode.

@BeateJ What do you think?

schodet, 2024-04-11:
Hi, I am the maintainer of _NXT Improved firmware_. The bug is fixed with version 1.29.6.

But the NXT Enhanced Firmware, which is quite popular, is no longer maintained, so I think that having the @Andy1978 fix would be valuable.

Thanks.

bjost2s, 2024-10-25:
out of scope of our project

closed 2024-10-25 by bjost2s