OpenRoberta/openroberta-lab #1650: Python Code Generation - Initialize Variable with use of isPrime (or any other Generated Helper Method) not working

NiHoffmann, 2024-03-20:
**Describe the bug**
When using a Python-Robot, initializing a variable with the help of a function that needs a generated helper function the program will crash. One of these functions is the isPrime function.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a Python-Robot here Lego Spike
2. Add global Variable to NEPO-Program
3. Add isPrime with any number as input here 3 as Initiatal value
4. Add a show text on LED Matrix with the created variable
5. Upload the Program to the Robot
6. See error

**Expected behavior**
 "True" will be shows by the LED-Matrix.

**Screenshots**
[image]

**Reason for behaviour**
___item = ___isPrime(3) will be called before the ___isPrime() method was visitied by the python interpreted.

**Additional context**
To check weather or not a function is a generated Helper Function check the Source Code of the NEPO-program.

**Device Information**
 - Device Laptop
 - OS:Linux
 - Browser Chrome

NiHoffmann, 2024-09-04:
I have created a potential fix for this problem at feature/Issue1650. This fix includes creating logical sections for the process of generating source code. In addition to the generate prefix method, methods such as generateImport, generateHelperMethods, .... will be added. This can be used to more easyly stucture the Program generate into different Parts. The Abstract logic on how a specific language will generate Code will be moved as far up the inheritance structure as Possible. Idealy every Programming Language will only once define this generation process. 
So far this change has only been done for the Python-Code-Generation, all other Visitors have been kept unchanged.

These also resulted in changes to Code generation that does not directly affect this, since the more unified approach also results in a more unified generation of empty lines , spacing.

It has been suggested to also adjust the other Visitors, although they do not suffer from this bug. It would still be nice to introduce this more structured approach for the code generation.
TODO:
- [x] Abstract-Language-Visitor
- [x] Python-Visitors
- [ ] CPP-Visitors
- [ ] Aseba-Visitors
- [ ] Java-Visitors
  
Instructions on how this process might look like have been anotated with "//TODO 1650 ..." as comments in the source code.

bjost2s, 2024-10-23:
works as expected

bjost2s, 2024-10-28:
forgot to check naoPythonSimVisitor -> not implemented

rbudde, 2024-10-29:
works as expected, including the NAO sim :-)

closed 2024-10-29 by rbudde