OpenRoberta/openroberta-lab #1657: EV3: faulty simulation with non-standard robot configuration

charlie-chaperon, 2024-04-16:
**Describe the bug**
When using a custom robot configuration by reversing left and right for motors B and C, the simulation does not work properly.
When simulating a line follower with a standard configuration, it works well. Reversing Left and right makes the simulated robot drive in a straight line and not reacting to any input from the light sensor. However the same program just works as expected when uploaded to a EV3.

standard robot configuration
[image]

custom robot configuration with left and right reversed 
[image]

Test program used: line following.
[image]

**Steps to reproduce the behavior:**

1. write a simple line follow program for the EV3 c4ev3 (see picture)
2. use standard robot configuration
3. simulate program - everything should work

4. switch sides of the motors from left/right (see picture)
5. simulate the new program - the robot drives in a straight line and does not react to sensor input

**Expected behavior**
The custom configuration should work like the standard configuration, just with the robot turning to the right instead of the left when not finding a line.

**Device information**
 - tested on several Laptops / different OS (Windows10, several debian-based Linux distributions)
 - tested Browsers: recent Chrome, Firefox, Brave - I tested various Browser versions of these, the problem appeared on everyone of them since I tested
 - Open Roberta versions: 5.2.4 to 5.2.14 - first notice of the issue: **august 2023**

**Additional context**
The programs do work on a real EV3, just the simulation seems bugged.

bjost2s, 2024-07-01:
Thanks for this issue, your enhancement makes sense, we will considerate this in a future version. Currently there is  no time to solve it soon.

bjost2s, 2024-07-02:
Solution should solve #1507, too