OpenRoberta/openroberta-lab #1673: Refactor Null Constant Usage in Blockly Logic Blocks and Test

alvaro555, 2024-07-10:
**Describe the bug**
The null constant was previously used inappropriately in the Integration test, specifically in listOperationsBasic the  block "in list X find first occurrence of item Y" (IndexOfFunct) . Although this specific issue has been resolved, there is a need to move the null constant from logic blocks and change its color to pink to correctly represent connection types. Additionally, the null constant should be removed from any tests that do not involve connection types or are used incorrectly.

**Additional context**
This issue was discovered because of the TypecheckCommonLanguageVisitor, which currently works with microbitv2, ev3, and WeDo.

bjost2s, 2024-07-11:
The null constant is more widely usable, here are the definition of possible types (from blockly, not only "connection"): 
 "output": ["Connection", "String", "Array_Boolean", "Array_Colour", "Array_Connection", "Array_Number", "Array_String"],
 
and here is a full list of toolboxes using it: 
[image]
@alvaro555, let's discuss the next steps together

rbudde, 2024-07-12:
btw: null is used inconsistently in blockly. You may create an array with null, but you cant update the array with null. The block which refers to the lists name ("l") jumps out of the set block. Assigning a string does work.
[image]