OpenRoberta/openroberta-lab #1677: NXT sim: different execution of the show text block in simulation and NXT

AHberta, 2024-07-14:
**Describe the bug**
 - Show text block shows a different behavior in the simulation than on the NXT: 
     NXT: Text output without first clearing the screen
     Sim: Clearing screen and then text output

**To Reproduce**
Steps to reproduce the behavior:
1. Import program issue_show-text.xml
2. Click on SIM to open the simulation
3. Click on NXT to display NXT brick
4. Start program
5. See the line-by-line output, whereby the previous lines are deleted again after shown

**Expected behavior**
 - Displays a line-by-line output, whereby the previous lines remain shown - so the behaviour on NXT

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome, Edge

Program: 
[issue_show-text.txt](https://github.com/user-attachments/files/16228078/issue_show-text.txt)

(txt = xml!)

AyaMoussa, 2024-10-01:
Issue solved on branch [feature/issue1677](https://github.com/OpenRoberta/openroberta-lab/tree/feature/issue1677)

AyaMoussa, 2024-10-22:
Issue solved. However, another issue persists. When more than one text is written in the same row, the new text is written over the old one, without deleting the old line

bjost2s, 2024-10-23:
NXT overlays in Pixel, so the behaviour is as expected

closed 2024-10-23 by bjost2s