OpenRoberta/openroberta-lab #1678: Calliope mini V3 - Neopixel RGB colour codes generate error message

PetraSohnius, 2024-07-15:
Describe the bug

- Input values for RGB LEDs of Calliope mini V3 generate folowing error message on the board: 
" :( Line 19 typeError function takes 2 positional arguments but 5 were given"

**To Reproduce**
Steps to reproduce the behavior:
1. insert "turn RGB LED on - all - colour"
2. connect with "colour with - red - green - blue - white"
3. type any value from 0 to 255 for the four colour options
4. send code to board
5. see error message

**Expected behavior**
 - Colour of RGB LEDs should be determined by numbers inserted. Simulation shows working LEDs.

**Device information**
- Type: Desktop
- OS: Windows 10
- Browser: Edge

[image]
[image]

AyaMoussa, 2024-07-19:
fixed on branch [Issue1678](https://github.com/OpenRoberta/openroberta-lab/tree/feature/Issue1678)

AHberta, 2024-07-25:
Tested and works perfectly! Thanx ;)

closed 2024-07-25 by bjost2s