OpenRoberta/openroberta-lab #1680: Open Roberta Lab Tour: Server error starting the interactive tour

AHberta, 2024-07-16:
**Describe the bug**
 - When attempting to start the interactive tour in Open Roberta Lab using the "Take an interactive Tour" button, a server error occurs. This error prevents the tour from loading and starting. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the start screen of Open Roberta Labs
2. Click on "Take an interactive tour" to start the tour
3. A popup with a server error is displayed

**Expected behavior**
 - It is expected that the start screen will be exited, switched to the Open Roberta Lab, and the tour will interactively explain the principles of program creation and testing in the simulation.

**Screenshots**
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome, Edge

AHberta, 2024-07-24:
Tested and works perfectly! Thanx ;)

bjost2s, 2024-07-31:
There is a very similar issue for working with old links to a program.

rbudde, 2024-07-31:
reviewed

closed 2024-07-31 by rbudde