OpenRoberta/openroberta-lab #1684: Tabs for Open Roberta xNN are not loaded when using a direct link

al-emonds, 2024-08-02:
**Describe the bug**
 - The tabs "NEURONALES NETZ Definieren" und "NEURONALES NETZ Lernen" are not displayed when the lab is opened by a direct link: [lab.open-roberta.org/?loadSystem=xNN](https://lab.open-roberta.org/?loadSystem=xNN).

**To Reproduce**
Steps to reproduce the behavior:
Put the direct link [lab.open-roberta.org/?loadSystem=xNN](https://lab.open-roberta.org/?loadSystem=xNN) into the browser (I tested Firefox, Chrome and Microsoft Edge on Windows.).

**Expected behavior**
 - The tabs "NEURONALES NETZ Definieren" und "NEURONALES NETZ Lernen" should be displayed when using the above link.

**Screenshots**
 
[image]

**Device information**
 - Type Desktop 
 - OS: Windows 10 (Version 22H2)
 - Browser: Firefox Version 115.13.0esr (64-Bit) and Google Chrome Version 127.0.6533.89 (Offizieller Build) (64-Bit) and Microsoft Edge Version 127.0.2651.74 (Offizielles Build) (64-Bit)

**Additional context**
 - Might be related to this issue https://github.com/OpenRoberta/openroberta-lab/issues/1679

bjost2s, 2024-10-26:
Please use from now on the generated link from the startview (click on the star of the desired system), e.g.

 https://dev.open-roberta.org/?loadSystem=xNN&extensions=nn

it works as expected.

bjost2s, 2024-11-04:
works as expected

closed 2024-11-04 by bjost2s