OpenRoberta/openroberta-lab #1697: Neural Network Disappears When Adding a Neuron After Running Forward Propagation

alvaro555, 2024-10-16:
### **Description:**

When running forward propagation in the Neural Network Define tab and then attempting to add or remove a neuron, the entire neural network disappears, leaving only a single neuron visible.

### **Steps to Reproduce:**
1. Open Open Roberta Lab.
2. Go to the **Neural Network Define** tab.
3. Set up a neural network.
4. Click the button **"Run forward propagation neuron by neuron"** (located in the bottom-left corner).
5. After clicking the "+" or "-" button to add or remove a neuron in the neural network:

   - **Expected Behavior**: The neural network should maintain its structure, and a new neuron should be added or removed from the appropriate layer without affecting the rest of the network.
   - **Actual Behavior**: The entire neural network disappears, and only a single neuron remains visible in the workspace.

---

### **Environment:**
- Browser: Chrome Version 129.0.6668.70 (Official Build) (64-bit)
- Operating System: Ubuntu 22.04.5 LTS

---

### **Screenshots/Logs:**
[image]

AyaMoussa, 2024-11-13:
fixed can be reviewed on branch [feature/Issue1697](https://github.com/OpenRoberta/openroberta-lab/tree/feature/Issue1697)

rbudde, 2024-12-12:
works as expected. Merged to develop

closed 2024-12-12 by rbudde