OpenRoberta/openroberta-lab #1703: Change MicroPython version for Calliope mini 3 to the current one.

joernalraun, 2024-11-06:
The current used MicroPython version does not support the standard way to start BLE mode (triple reset) on Calliope mini 3.
If you update to the current version it will work:
(https://github.com/calliope-edu/micropython-calliope-mini-v3/tree/oct24-mini3)

joernalraun, 2024-11-21:
I just bump this as people are flooding support because of this issue and it should be quick to implement...

Cruiser79, 2024-12-12:
Hello @joernalraun 

is this the same as this PR on the other repo https://github.com/OpenRoberta/ora-cc-rsc/pull/15 ?
Than i am also interested in this fix.

joernalraun, 2024-12-12:
This is the same Hex-file, yes! However, the link in this comment is pointing to the generated file from Calliope, the PR has the file already integrated in the lab (so even less work to do)... If you are one of the few how was able to host this on your own machine, you just can switch the Hex-file or merge this change to your current working branch. Please tell me, if there are any issues with it (though it should work well, as this is running for a while at python.calliope.cc).

rbudde, 2024-12-12:
btw: we have wo different hex-binaries for the v3 in the ora-cc-rsc repo:
1. one in directory firmware (this is for the "factory reset" command
2. one in directory runtimeHex (this is the basis to put your program into)
I'll prepare a commit that will change both to the new version today. I'll deploy this on one of out test servers, so it can be tested by everyone. I'll write a comment which I've taken later. Stay tuned :-)

Cruiser79, 2024-12-12:
@rbudde nice to hear about that. Can you tell me, how an uninformed user like me can reach this test servers :-)

rbudde, 2024-12-12:
@joernalraun I took your hex-file and put it into the directory runtimeHex. Then I downloaded a copy of the predefined program ("factory reset") into directory firmware.

I deployed this on dev3

- the v3 works as expected with usb from my laptop using chrome
- using the calliope mini app on my Android tablet and setting the editor to dev3
  - pairing works
  - uploading works
  - the program runs, BUT
  - when uploading the next time the pairing has to be repeated (then it works again)
  - resetting the v3 to factory default works (with bluetooth)
  - then the next program can be uploaded (i.e. the pairing is not lost in this case)
  - but then the pairing is lost and must be done again

There must be a difference i the pairing logic between the predefined program ("factory reset") and the hex-file to which the user program is attached to. Or?

I have no IOS device. @AHberta or @BeateJ  can you test the v3 with your IOS devices with dev3?

@Cruiser79 all out test systems are public, but of course should be used for testing mainly :-):
- https://test.open-roberta.org/ is our nighly build
- https://dev.open-roberta.org/ is our snapshot system (updated after every commit to **develop**)

- https://dev3.open-roberta.org/ got our changes w.r.t. the Calliope V3

joernalraun, 2024-12-12:
@rbudde great!
The behavior seems to be correct, since as of now radio and Bluetooth communication can't run in parallel on MicroPython and radio is always active, so you will need to manually wake up the Bluetooth communication by hitting 3x reset. The pairing has to be done only once, but the mode to wait for data has to be initiated. The factory program is no MicroPython program, so this works without the need to initiate the Bluetooth state (same in MakeCode etc.).

Cruiser79, 2024-12-20:
@rbudde I tried the dev3 version but can not upload a program to my calliope V3. I tried it with my laptop, my Android 15 Smartphone and with my old Android 8.1 Tablet directly on the website. On every device i paired the calliope, paring was successfully. But when i try to upload a program, in the popup window "dev3.open-roberta.org try to make a connection" (translated from german) there are no compatible devices.
If i connect the calliope over usb, the upload is working and the "dev3.open-roberta.org try to make a connection" window is filled with my calliope.
Also i tried it with the calliope mini app on my Android 15 and on the Android 8.1 device and get another error on that ("„Achtung. Cannot read properties of undefined (reading ‚requestDevice‘)") The window with the compatible devices is not showing here. All rights for the app are allowed.
Why is it working on your device, but not on my? Why don't you get this "requestDevice" error, but me on both devices?

Cruiser79, 2024-12-20:
> @Cruiser79 you cannot use Bluetooth from your web browser to transfer the file. You do need to use the app (and from there use Open Roberta as a webview). If you want to make a direct copy from Android, you will need to have an OTG cable (the original Calliope mini V3 cable should be sufficient) and an app that can copy and select files from and to your device. On your laptop web USB or a drag'n'drop of the Hex-file to the "MINI" drive should work immediately.

Ok, i understand that. But then my second problem still remains:

Also i tried it with the calliope mini app on my Android 15 and on the Android 8.1 device and get another error on that ("„Achtung. Cannot read properties of undefined (reading ‚requestDevice‘)")

So why is this error showing on my devices?

joernalraun, 2024-12-20:
That is indeed an error from Open Roberta that is separated from the app, so maybe @rbudde or @BeateJ can answer?