OpenRoberta/openroberta-lab #1713: Unintended Execution of Commands Despite 'If' Conditions in Robot Program

Lexa2000, 2024-12-06:
When you run the program, EVERY command is executed (regardless of whether the "if" condition is met). And I don't think that should be the case. You can see this if you test it in debug mode and place the robot on the white area. To test whether it's the program or the lab, I tested a simpler program. And there it is also the case that the robot always turns, although both sensors output "Wrong". What you can observe is not what it says or expects.

Lexa2000, 2024-12-06:
https://github.com/user-attachments/assets/f4cb882d-cf69-4573-8aed-1c389a8444f6

[image]

Lexa2000, 2024-12-06:
[Mbot-Suiveur-ligne-Correction(1).zip](https://github.com/user-attachments/files/18037864/Mbot-Suiveur-ligne-Correction.1.zip)

profedetecnologia, 2025-01-13:
Having the same issue here. Is there some workaround? We use mbot in our classroom and I was just testing some simulatios with the same results.
Thanks!

NiHoffmann, 2025-01-22:
This Issue has been Fixed here: [Issue1713](https://github.com/OpenRoberta/openroberta-lab/tree/feature/Issue1713)

bjost2s, 2025-02-13:
tested, works fine

closed 2025-02-13 by bjost2s